erfanoabdi/ErfanGSIs #10: missing /makeroot.sh makeroot.sh makea.sh etc

hpxing, 2019-06-12:
i tried to make coloros gsi
./make.sh: 行 141: /home/steven/ErfanGSIs/prebuilt/9/makeroot.sh: 没有那个文件或目录

$romsdir/$sourcever/$romtype/makeroot.sh "$systemdir"

cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-26-arm32/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-26-arm64/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-sp-26-arm32/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-sp-26-arm64/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-27-arm64/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-sp-27-arm32/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-sp-27-arm64/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-sp-28-arm32/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/vndk-sp-28-arm64/*' 的文件状态(stat): 没有那个文件或目录
cp: 无法获取'/home/steven/ErfanGSIs/prebuilt/vendor_vndk/moreP/lib64/*' 的文件状态(stat): 没有那个文件或目录

hpxing, 2019-06-12:
Creating Image: ColorOS-AB-9-20190612-ErfanGSI.img
Prepare File Contexts
/home/steven/ErfanGSIs/scripts/../tmp/file_contexts: Multiple same specifications for /system/bin/vndk-detect.
/home/steven/ErfanGSIs/scripts/../tmp/file_contexts: Multiple same specifications for /system/etc/usb_audio_policy_configuration.xml.
/home/steven/ErfanGSIs/scripts/../tmp/file_contexts: Multiple same specifications for /system/bin/rw-system.sh.
/home/steven/ErfanGSIs/scripts/../tmp/file_contexts: Multiple same specifications for /system/bin/phh-on-boot.sh.
/home/steven/ErfanGSIs/scripts/../tmp/file_contexts: Multiple same specifications for /system/bin/wificonf.
/home/steven/ErfanGSIs/scripts/../tmp/file_contexts: Invalid argument
Remove Temp dir

MacTavishAO, 2019-06-12:
Multiple same specifications for /system/bin/vndk-detect. I guess you are trying to rebuild gsi over gsi. Remve these lines from mounted system's etc/selinux/plat_file_contexts.

erfanoabdi, 2019-06-13:
no, just remove tmp folder

MacTavishAO, 2019-06-15:
Did your problem fixed ? I'll close issue.

closed 2019-06-28 by erfanoabdi