erfanoabdi/ErfanGSIs #11: Android 9 <montana> [Bootloop on animation] - Latest OxygenOS

JarlPenguin, 2019-06-12:
**Describe the bug**
The OxygenOS GSI bootloops.

**To Reproduce**
Steps to reproduce the behavior:
1. Install LineageOS 16  (Treble)
2. Format system and install OOS GSI.
3. Reboot

**Expected behavior**
The GSI boots properly.

**Screenshots**
N/A

**Logs**
Add logs to help your problem.
Log file: https://del.dog/uvopiyibug.pl

**Smartphone (please complete the following information):**
 - Device: montana
 - Vendor: LineageOS 16.0
 - Version Pie

**Additional context**
Add any other context about the problem here.

erfanoabdi, 2019-06-12:
remove com.qualcomm.location.xml form your system/etc/permissions or vendor/etc/permissions
that xml shouldn't be on vendor anyway

JarlPenguin, 2019-06-13:
Doing that fixed it

closed 2019-06-13 by JarlPenguin