erfanoabdi/ErfanGSIs #8: MAC install error No available formula with the name "liblzma-dev"

hpxing, 2019-06-12:
Brew install protobuf liblzma-dev brotli lz4
Error: No available formula with the name "liblzma-dev" 
lzma is now part of the xz formula:
  brew install xz

should i changed to  Brew install protobuf xz brotli lz4

erfanoabdi, 2019-06-12:
yep do it plz

hpxing, 2019-06-12:
Extracting archive: downloads/temp.zip
--
Path = downloads/temp.zip
Type = zip
Physical Size = 2044566177

Everything is Ok 

Size:       3758096384
Compressed: 2044566177
Invalid sparse file format at header magi
Failed to read sparse file
./zip2img.sh: line 200: gawk: command not found

how to solve this?

spdroid, 2019-06-12:
@hpxing https://github.com/erfanoabdi/ErfanGSIs/issues/8#issuecomment-501247143

hpxing, 2019-06-12:
mount: You must specify a filesystem type with -t.  
macos can not mount ext4 
i am now using ubuntu to try in vm

MacTavishAO, 2019-06-12:
> mount: You must specify a filesystem type with -t.
> macos can not mount ext4
> i am now using ubuntu to try in vm

use [fuse-ext2](https://github.com/alperakcan/fuse-ext2)

closed 2019-07-20 by erfanoabdi