erfanoabdi/ErfanGSIs #12: <Pie> <Mido> Gsi building no such file/dir errors

hjyoung1, 2019-06-13:
**Describe the bug**
Building oxygen os gsi gives bunch of  "no such file or directory found errors"
Even after errors gsi build was successful but there's problem with booting. 

**To Reproduce**

Followed Readme.md file using ./url2GSI *op6 firmware link here* OxygenOS 

**Screenshots**

[image]

**Logs**
[log.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3287833/log.txt)

**Smartphone (please complete the following information):**
 - Device: Mido
 - Vendor: Revenge OS v4, v7, los16 
 - Version: Pie

**Additional context**

Gsi boots for 1 sec then reboot again. Endless loop. 
Also above listed vendors boot op7pro gsi perfectly. And other gsis too.

MacTavishAO, 2019-06-14:
Hey, seems like the problem is `#00 pc 00000000000dcc4c  /system/lib64/libsurfaceflinger.so (android::ExBufferLayer::isHDRLayer() const+48)` if you can give me system/lib64/libsurfaceflinger from op6 i can fix this.

hjyoung1, 2019-06-14:
Thanks you already fixed that on telegram xD boots fine now. Only round corners bug now but that's not related here. So closing this, Thanks.

closed 2019-06-14 by hjyoung1

NZXT03, 2019-06-14:
That's OK. Just install gawk.