erfanoabdi/ErfanGSIs #22: <Android 9 Pie> <Vince> [Screen Recorder Not Working]

oyiramname, 2019-06-29:
**Screen Recorder Not Working**
Screen Recorder Automatically disepear/closed when after touch Record Button.

**Screenshots**
[image]

**Logs**
[Log]([Log](https://drive.google.com/file/d/1INQCwRfEVtNJaGx9ogH2uUQHSjiqKBs6/view?usp=drivesdk) 

**Smartphone 
 - Device : Redmi 5 Plus A.k.a Vince
 - Vendor : Serein OS Pie (AOSP)
 - Version : Pie

**Additional context**
Please @erfanoabdi Fix Screen Recorder Issue. Please... please... please...

Telegram : @oyiramname

oyiramname, 2019-06-29:
This is on OxygenOS A Only 9 20190629 ErfanGSI

MacTavishAO, 2019-09-20:
This issue is has been dead for couple of months. Use another screen recorder. Closed.

closed 2019-09-20 by MacTavishAO