erfanoabdi/ErfanGSIs #25: Q

NZXT03, 2019-07-11:
How to Port pixel developer previews to Q gsi,
If we cannot mount them? Any instructions or hints available? Should I convert walleye's system.img to raw, and only after this step mount with mount point?

closed 2019-07-13 by NZXT03

UP5468-GE-ES44AC, 2019-09-14:
Did you find out a solution?I cannot mount system.img (pixel's firmware)too

triskel777, 2019-09-15:
change your twrp recovery
use latest orange fox recovery instead

UP5468-GE-ES44AC, 2019-09-16:
I found out a solution:
this command can mount a pixel's system.img
mount -t auto -o ro,loop cache/system.img system
[image]

UP5468-GE-ES44AC, 2019-09-16:
> I found out a solution:
> this command can mount a pixel's system.img
> mount -t auto -o ro,loop cache/system.img system
> [image]

but I always meet this error：
[image]

NZXT03, 2019-09-16:
Which firmware you are building from?