erfanoabdi/ErfanGSIs #26: Pixel-AB-10-20190711-GSI OnePlus6 (Enchilada) [Bootloop on bootanimation (system services crash)]

vijai1996, 2019-07-14:
**Description:**
Flashed Pixel-AB-10-20190711-ErfanGS on Oneplus 6 with latest stock Oxygen OS flashed on both slots. Rebooted and stuck at boot animation. Logcat (attached) says it couldn't find partition "config" by name and subsequently all services is stopped and the boot process is repeated again.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Pixel-AB-10-20190711-ErfanGSI.img
2. Flash permisiver-v4
3. reboot

**Expected behavior**
Boot into device setup

**Log:**
[erfan-Pixel-AB-OP6-bootlog.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3389505/erfan-Pixel-AB-OP6-bootlog.txt)

**Smartphone:**
 - Device: Oneplus 6 Enchilada
 - Vendor: Oxygen OS 9.0.7 vendor
 - Version: Pie

erfanoabdi, 2019-07-14:
thanks for testing on op6
open your system/vendor/build.prop
and remove this line:
`ro.frp.pst=/dev/block/bootdevice/by-name/config`

vijai1996, 2019-07-14:
Thanks. That worked

closed 2019-07-14 by vijai1996