erfanoabdi/ErfanGSIs #28: /home/parallels/ErfanGSIs/scripts/mkimage.sh: line 94: 26065 Floating point exceptionsudo $make_ext4fs -T 0 -S $fcontexts -l $syssize -L / -a / -s "$output" "$systemdir/" Remove Temp dir

hpxing, 2019-07-18:
**Describe the bug**
A clear and concise description of what the bug is.
when i build gsi it is said /home/parallels/ErfanGSIs/scripts/mkimage.sh: line 94: 26065 Floating point exceptionsudo $make_ext4fs -T 0 -S $fcontexts -l $syssize -L / -a / -s "$output" "$systemdir/"
Remove Temp dir

if it manualy set the syssize the script works ok

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

hpxing, 2019-07-18:
# Getting system size and add approximately 5% on it just for free space
systemsize=`du -sk $systemdir | awk '{$1*=1024;$1=int($1*1.05);printf("%d\n",$1)}'`
echo $1 $systemsize

this works

closed 2019-07-18 by hpxing

balliboxer, 2019-07-18:
@hpxing To solve this issue just type in cmd
1 = Mkdir out
2 = mkimage.sh systemmount OxygenOS AA out
U have to guide the mkimage.sh to out folder at end

hpxing, 2019-07-18:
> @hpxing To solve this issue just type in cmd
> 1 = Mkdir out
> 2 = mkimage.sh systemmount OxygenOS AA out
> U have to guide the mkimage.sh to out folder at end
not this reason》 because the syssize is Scientific Counting  like 11111e5
the  $make_ext4fs  accept number like 123444555 integer