erfanoabdi/ErfanGSIs #29: <PIE> <Xperia X Performance> [Animation]

sjllls, 2019-07-19:
Freeze on Boot Animation.
ZenUI-Aonly-9-20190719
OxygenOS-Aonly-9-20190701
[ZenUI-Aonly-9-20190719.log](https://github.com/erfanoabdi/ErfanGSIs/files/3409240/ZenUI-Aonly-9-20190719.log)
[OxygenOS-Aonly-9-20190701.log](https://github.com/erfanoabdi/ErfanGSIs/files/3409241/OxygenOS-Aonly-9-20190701.log)

rsyhan, 2019-07-19:
For OxygenOS:
03-10 09:43:57.358  3041  3041 F DEBUG   : *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
03-10 09:43:57.359  3041  3041 F DEBUG   : Build fingerprint: 'OnePlus/OnePlus6/OnePlus6:9/PKQ1.180716.001/1905251800:user/release-keys'
03-10 09:43:57.359  3041  3041 F DEBUG   : Revision: '0'
03-10 09:43:57.359  3041  3041 F DEBUG   : ABI: 'arm64'
03-10 09:43:57.359  3041  3041 F DEBUG   : pid: 2974, tid: 2974, name: surfaceflinger  >>> /system/bin/surfaceflinger <<<
03-10 09:43:57.359  3041  3041 F DEBUG   : signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x90000000000
03-10 09:43:57.359  3041  3041 F DEBUG   :     x0  0000007c3ee4d610  x1  0000007c3c9ac980  x2  0000007c3c800000  x3  0000000000000004
03-10 09:43:57.359  3041  3041 F DEBUG   :     x4  00000000000001ac  x5  0000000000000004  x6  0000007c3ee2bac8  x7  0000000000000001
03-10 09:43:57.359  3041  3041 F DEBUG   :     x8  1000090000000000  x9  0000000000000002  x10 0000000000000001  x11 0000000000000000
03-10 09:43:57.359  3041  3041 F DEBUG   :     x12 00000000000000b4  x13 00000000000000b5  x14 0000007c3e1ea0d0  x15 0000007c3e1ebee8
03-10 09:43:57.359  3041  3041 F DEBUG   :     x16 0000007cbf9764f0  x17 0000007cc04abe74  x18 354d5a0696eabcb0  x19 0000007c3be78000
03-10 09:43:57.359  3041  3041 F DEBUG   :     x20 0000007fddf97798  x21 0000007fddf977c0  x22 0000000000000000  x23 0000000000000001
03-10 09:43:57.359  3041  3041 F DEBUG   :     x24 0000000000000008  x25 0000007cc1a315e0  x26 0000007cc08fc130  x27 0000007cc08fc068
03-10 09:43:57.359  3041  3041 F DEBUG   :     x28 0000000000000001  x29 0000007fddf97590
03-10 09:43:57.359  3041  3041 F DEBUG   :     sp  0000007fddf97580  lr  0000007cbf8c4c38  pc  0000007cbf8c4c4c
03-10 09:43:57.370  3041  3041 F DEBUG   : 
03-10 09:43:57.370  3041  3041 F DEBUG   : backtrace:
03-10 09:43:57.370  3041  3041 F DEBUG   :     #00 pc 00000000000dcc4c  /system/lib64/libsurfaceflinger.so (android::ExBufferLayer::isHDRLayer() const+48)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #01 pc 0000000000067714  /system/lib64/libsurfaceflinger.so (android::BufferLayer::onDraw(android::RenderArea const&, android::Region const&, bool) const+200)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #02 pc 00000000000b31fc  /system/lib64/libsurfaceflinger.so (android::SurfaceFlinger::doComposeSurfaces(android::sp<android::DisplayDevice const> const&)+1544)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #03 pc 00000000000b02ec  /system/lib64/libsurfaceflinger.so (android::SurfaceFlinger::handleMessageRefresh()+1896)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #04 pc 00000000000afae0  /system/lib64/libsurfaceflinger.so (android::SurfaceFlinger::onMessageReceived(int)+3836)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #05 pc 0000000000013fbc  /system/lib64/libutils.so (android::Looper::pollInner(int)+336)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #06 pc 0000000000013dd0  /system/lib64/libutils.so (android::Looper::pollOnce(int, int*, int*, void**)+60)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #07 pc 000000000009fef0  /system/lib64/libsurfaceflinger.so (android::impl::MessageQueue::waitMessage()+84)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #08 pc 00000000000ae374  /system/lib64/libsurfaceflinger.so (android::SurfaceFlinger::run()+20)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #09 pc 0000000000003220  /system/bin/surfaceflinger (main+932)
03-10 09:43:57.370  3041  3041 F DEBUG   :     #10 pc 00000000000acdac  /system/lib64/libc.so (__libc_init+88)

surfaceflinger crashed because of isHDRLayer.You should flash the patch here:
[OOSHDRfixAonly.zip](https://github.com/erfanoabdi/ErfanGSIs/files/3409581/OOSHDRfixAonly.zip)

rsyhan, 2019-07-19:
For ZenUI I can't exactly to say what problem it is but I guess it also caused hasHDRDisplay:
```
03-10 15:40:39.876   739   739 I chatty  : uid=1000(system) /system/bin/surfaceflinger identical 1 line
03-10 15:40:39.926   739   739 I ConfigStore: android::hardware::configstore::V1_0::ISurfaceFlingerConfigs::hasHDRDisplay retrieved: 0
03-10 15:40:39.935   791   884 I ServiceManager: Waiting for service 'package_native' on '/dev/binder'...
```
so have a try yourself to fix HDR Display:
https://github.com/erfanoabdi/ErfanGSIs/blob/master/docs/HDRfix.md

sjllls, 2019-07-19:
OSS Solved, thanks a lot.

closed 2019-07-19 by sjllls