erfanoabdi/ErfanGSIs #30: <Pixel Q GSI> <Tulip> [Bootloop on animation]

BrkeMem, 2019-07-19:
Bootloop on the boot animation.

Steps to reproduce the behavior:
1. Flash system
2. Flash FixZygote
3. Flash Permissiver
4. Reboot

Log:
[log.zip](https://github.com/erfanoabdi/ErfanGSIs/files/3409449/log.zip)

Smartphone:
 - Device: Tulip (Redmi Note 6 Pro)
 - Vendor: MIUI Pro
 - Version Pie

Additional context
For some reason an old version of the Pixel Q GSI booted fine with the same vendor

MacTavishAO, 2019-07-20:
W ServiceManagement: Waited one second for android.frameworks.sensorservice@1.0::ISensorManager/default. Waiting another...

Try different vendor.

erfanoabdi, 2019-07-20:
install zygote fix v2 : https://androidfilehost.com/?fid=1395089523397961748

MacTavishAO, 2019-09-20:
Zygotefix v2 should've fixed it. But no report back from issue creator. Closed.

closed 2019-09-20 by MacTavishAO