erfanoabdi/ErfanGSIs #34: <Pixel Q GSI> <tissot> [Incall sound issue/ Google Camera Issue/ Lockscreen Issue]

zurabi97, 2019-07-20:
**Describe the bug**
During call, I cannot hear what other say and the other person don't hear me.
On Google camera there's black screen (I give permissions for it in Settings but nothing)
When I finish to set up a pattern/pin/password, pop up a toast message that there's already setup a password, but it's not.

**To Reproduce**
Steps to reproduce the behavior:
Incall:
1. Calling a number.
2. No one hear our voices.

Camera:
1. Opening camera app.
2. Black screen.

1. Setting up a lockscreen on security settings.
2. Toast message pops that there's already a password.
3. No password set.

**Expected behavior**
1. No Sound on incall.
2. Black Screen in Google Camera.
3. Refuse to set a password

**Screenshots**
-

**Logs**
[log.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3413987/log.txt)

**Smartphone (please complete the following information):**
 - Device: Mi A1
 - Vendor: Pie (by flex1911)
 - Version: Pixel Q GSI/ Generic Q

**Additional context**

MacTavishAO, 2019-07-20:
1- finish setupwizard. 2- it's not our problem try enabling hal3. 3- What ?

zurabi97, 2019-07-20:
1. Reinstalled gsi and trying again. 2. OK. 3. It's fixed somehow by reflashing the gsi.

zurabi97, 2019-07-21:
1. incall fixed, 2. Enabling HAL3 doesn't work. 3. Lockscreen fixed.
Bonus: Bluetooth is broken but I guess it's due to "persist" with missing Bluetooth components

MacTavishAO, 2019-07-21:
Can you install Open Camera app and test camera2api ? If it works it's gcam problem.

zurabi97, 2019-07-21:
It's Google Camera's problem, I'll try a fix in the group about that. As about custom kernels, I will try if the developer didn't add to AnyKernel3 zip the Android Version check

closed 2019-09-20 by MacTavishAO