erfanoabdi/ErfanGSIs #35: <Android version> <Device code name> [Bug name Ex. Bootloop on animation]

SakshamGoyal, 2019-07-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

SakshamGoyal, 2019-07-21:
My phone boots then reboot again and again

corsicanu, 2019-07-21:
@SakshamGoyal Hard to follow the bug report draft?

alexa-v2, 2019-07-21:
@SakshamGoyal Did you even try to follow the format and fill in the information?

MacTavishAO, 2019-07-21:
Fill the information.

closed 2019-07-21 by MacTavishAO

SakshamGoyal, 2019-07-21:
Device Moto g5s plus Sanders
Vender per caf

SakshamGoyal, 2019-07-21:
I am facing bootloop on my device Sanders vender per car version 9