erfanoabdi/ErfanGSIs #36: <9.0> <violet> [Bootloop on animation] - Flyme

DiseR-LawnX, 2019-07-21:
**Describe the bug**
My device is stuck in the boot animation

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'recovery',clean all data
2. flash pe vendor,then flash gsi and permissiver
3. reboot
4. See error

**Expected behavior**
boot successfully

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
[logcat.log](https://github.com/erfanoabdi/ErfanGSIs/files/3414201/logcat.log)
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [violet]
 - Vendor: [PE CAF]
 - Version [Pie
]

**Additional context**
Add any other context about the problem here.

rsyhan, 2019-07-21:
[image]
I guess you are getting recovery log

Try to get system log from here : /data/local/tmp/logs/log

DiseR-LawnX, 2019-07-21:
I have to generated new logs
[recovery[1].log](https://github.com/erfanoabdi/ErfanGSIs/files/3414209/recovery.1.log)
[logcat[1].log](https://github.com/erfanoabdi/ErfanGSIs/files/3414210/logcat.1.log)

rsyhan, 2019-07-21:
> I have to generated new logs
> [recovery[1].log](https://github.com/erfanoabdi/ErfanGSIs/files/3414209/recovery.1.log)
> [logcat[1].log](https://github.com/erfanoabdi/ErfanGSIs/files/3414210/logcat.1.log)

Both are recovery log again.....Why not follow the file I had told you?

DiseR-LawnX, 2019-07-21:
原来是国人吗,才发现
我/data/local/tmp/logs/log这个目录下没有文件啊

DiseR-LawnX, 2019-07-21:
我是用的rec生成的log
包括
[image]
这几个文件

closed 2019-07-24 by DiseR-LawnX