erfanoabdi/ErfanGSIs #41: <9.0> <Tulip> [Bootloop on animation] Oxygen OS

arturo51236, 2019-07-27:
[log.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3438289/log.1.txt)

arturo51236, 2019-07-27:
Bootloop on the boot animation.

Steps to reproduce the behavior:

Flash system
Flash Permissiverv4
Flash FixZygotev2
Reboot

Log:
[log.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3438289/log.txt)

Smartphone:

Device: Tulip (Redmi Note 6 Pro)
Vendor: Xiaomi.eu
Version Pie

MacTavishAO, 2019-07-27:
You must not flash zygotefix on gsis based on android pie.

arturo51236, 2019-07-27:
Upss... Thanks, I've tried again and now boot but i have a black screen, new log:
[log[1].txt](https://github.com/erfanoabdi/ErfanGSIs/files/3438327/log.1.txt)

MacTavishAO, 2019-07-27:
Can you wipe data, remove /vendor/overlay and /system/app/treble-overlay-xiaomi-redminote6pro-systemui and try again ?

arturo51236, 2019-07-27:
/overlay and /treble-overlay-xiaomi-redminote6pro-systemui don't exists...

MacTavishAO, 2019-07-27:
tap to vendor and system from twrp's mount options.

arturo51236, 2019-07-27:
/overlay removed, the other still does not appear after mounting system, I tried to restart

MacTavishAO, 2019-07-27:
okay then wipe and reboot.

arturo51236, 2019-07-27:
Okey, it boot but i have this error, after touching "ok" i've got a gray screen, if i touch power button it gets a totally black.
[image]

Here is a new log:
[log[3].txt](https://github.com/erfanoabdi/ErfanGSIs/files/3438362/log.1.txt)

closed 2020-05-08 by arturo51236