erfanoabdi/ErfanGSIs #46: <ColorOS> <Mido> [Bootloop on animation]

hsuanhanSU, 2019-08-06:
**Describe the bug**
Bootloop on Animation

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Treble Rom(Revenge OS)
2. Flash ColorOS GSI system image (0323Aonly)
3. Flash Permissiver V2
4. Reboot system
5. See error

**Expected behavior**
Reboot to system

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
https://www.sendspace.com/file/wjgi3x

**Smartphone (please complete the following information):**
 - Device: Mido
 - Vendor: Revenge OS
 - Version Pie

**Additional context**
Add any other context about the problem here.

hsuanhanSU, 2019-08-06:
Update:
When I flash newer ColorOS GSI(0605)
It says"Size of image is larger than target device

spdroid, 2019-08-21:
Multiple thread for same issue. 

"Size of image is larger than target device" - it means the system img is bigger than your device's system size.

closed 2019-08-21 by spdroid