erfanoabdi/ErfanGSIs #51: <Android version> <Device code name> [Bug name Ex. Bootloop on animation]

rahil253655, 2019-08-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

rahil253655, 2019-08-17:
[recovery.log](https://github.com/erfanoabdi/ErfanGSIs/files/3512002/recovery.log)

Device- Realme X 
Vendor - stock
Version - pie

Stuck on bootanimation of oos

spdroid, 2019-08-21:
Rahil vaidhya, beta log recovery k nahi dene hai. Recovery logs ka kya achaar dalenge yaha.