erfanoabdi/ErfanGSIs #52: <miui Q MIUI-AB-10-20190813-ErfanGSI> <tissot> [Bug name Ex. Bootloop on animation]

hpxing, 2019-08-17:
**Describe the bug**
A clear and concise description of what the bug is.
Bootloop on animation] 

[log.zip](https://github.com/erfanoabdi/ErfanGSIs/files/3512305/log.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

erfanoabdi, 2019-08-17:
thanks for logs, yeah non pixel Q gsis is still WIP and buggy

MacTavishAO, 2019-08-17:
@hpxing Can't you just fill the form ?

closed 2019-08-21 by spdroid