erfanoabdi/ErfanGSIs #53: <Android version> <Device code name> [Bug name Ex. Bootloop on animation]

erkanbey1, 2019-08-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

erkanbey1, 2019-08-17:
<any Android Q gsi> <X00T> [i dont have fingerprint in settings ]

MacTavishAO, 2019-08-17:
Can u download [https://raw.githubusercontent.com/LineageOS/android_frameworks_native/lineage-16.0/data/etc/android.hardware.fingerprint.xml](url) this file and copy to vendor/etc/permissions ?

erkanbey1, 2019-08-17:
> Can u download
> https://raw.githubusercontent.com/LineageOS/android_frameworks_native/lineage-16.0/data/etc/android.hardware.fingerprint.xml
> <http://url> this file and copy to vendor/etc/permissions ?

Nasıl yapcam anlamadım .xml olarak mi indircem

şunu yazdı:

erkanbey1, 2019-08-17:
Zaten varmis ayni dosya

closed 2019-08-21 by spdroid