erfanoabdi/ErfanGSIs #54: <any Android q gsi> <X00T> [i have fingerprint.xml in vendor/etc/permissions but not in settings security]

erkanbey1, 2019-08-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

adi8900, 2019-08-18:
@erkanbey1 Where logcat?

amackpro, 2019-08-18:
@erkanbey1 Bad vendor .. as u Using  X00T  I recommend you to use Nitrogen Os latest vendor for base and Latest firmware

erkanbey1, 2019-08-18:
https://m.youtube.com/watch?v=By6dXXYjPjc i do same things but no fingerprint in settings

MacTavishAO, 2019-08-21:
Logs are most important thing in our lives.

MacTavishAO, 2019-09-02:
Lack of information and dead conversation. Closed.

closed 2019-09-02 by MacTavishAO