erfanoabdi/ErfanGSIs #58: <PIXEL Q GSI> <RMX1901> <Bootloop on animation>

sayamsamal, 2019-08-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start from ArrowOS GSI (Booted up)
2. Wipe System
3. Flash Android Q GSI (latest) Pixel-AB-10-20190821-ErfanGSI
4. Flash Permissiver_v5
5. Reboot (bootloop on bootanimation)
6. Format data
7. Reboot on (bootloop on animation)

**Expected behavior**
The device boots up

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

[log.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3530711/log.txt)

**Smartphone (please complete the following information):**
 - Device: Realme X
 - Vendor: Android GSI
 - Version [e.g. Pie] Android Q

**Additional context**
Add any other context about the problem here.
The GSI's working are Erfan's MIUI, ArrowOS, Descendant, AOSP by phh.

evilexecutable, 2019-10-23:
This almost looks like the logcat I get from a device-specific ROM on another device. Very interested to see if this gets fixed.

erfanoabdi, 2019-10-23:
'cannot setexeccon('u:r:rutilsdaemon:s0') for prop-adap-core: Invalid argument'

drop u:r:rutilsdaemon:s0 from your vendor sepolicy

erfanoabdi, 2019-10-23:
@erfanoabdi @evilexecutable isn't here?

erfanoabdi, 2019-10-23:
@erfanoabdi please join telegram group https://t.me/ErfanGSIs