erfanoabdi/ErfanGSIs #65: <Android version> <Device code name> [Bug name Ex. Bootloop on animation]

rajeshjaga, 2019-09-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

Adithya-Sekhar, 2019-09-07:
🙄

rajeshjaga, 2019-09-07:
Sorry i dint know

rajeshjaga, 2019-09-07:
how do i remove this

closed 2019-09-07 by rajeshjaga

Adithya-Sekhar, 2019-09-07:
you can't, just leave it