erfanoabdi/ErfanGSIs #69: gsi tool bug

hpxing, 2019-09-20:
root@hpxing-PC:/home/hpxing/ErfanGSIs# ./zip2img.sh input/6620_FIRMWARE.tgz 
Create Temp and out dir
Extracting firmware on: /home/hpxing/ErfanGSIs/cache
Aonly OTA detected
Extracting system
root@hpxing-PC:/home/hpxing/ErfanGSIs# mount cache/system.img  system
mount: mount point system does not exist
root@hpxing-PC:/home/hpxing/ErfanGSIs# mount cache/system.img  system
root@hpxing-PC:/home/hpxing/ErfanGSIs# ./make.sh  system Flyme Aonly output
Create Temp dir
Making copy of source rom to temp
Patching started...
Start Patching Light Services for Brightness Fix...
Create Temp dir

when i make gsi , the shell stuck at Start Patching Light Services for Brightness Fix..., howlong you wait it still this . but if you press ctrl+C to end it , it goes on . and build gsi works

itz63c, 2019-09-20:
@hpxing shut the fuck up

MacTavishAO, 2019-09-20:
Are you sure you run setup.sh ? And it will use cpu to patch it so, if you have one core that could be problem.

hpxing, 2019-09-24:
yes! i use parallels desktop and set cpu 2 cores!  i will try again

hpxing, 2019-09-26:
after i set cup 2 cores it works ～～
i think shoule ention on the readme  
close the issue

closed 2019-09-26 by hpxing