erfanoabdi/ErfanGSIs #88: <MIUI 11 - Android 10> <L78071> [camera don't work]

depauperado, 2019-10-13:
**Describe the bug**
camera don't work in the build MIUI-AB-10-20191012 on lenovo z5s.

**To Reproduce**
Steps to reproduce the behavior:
1. flash gsi miui on the phone
2. start phone
3. after configuration, open camera app
4. See error

**Expected behavior**
The camera app starts without errors.

**Screenshots**
[image]

**Logs**
[logcat_miui.log](https://github.com/erfanoabdi/ErfanGSIs/files/3721382/logcat_miui.log)

**Smartphone (please complete the following information):**
 - Device: [L78071]
 - Vendor: [lenovo]
 - Version [android 10]

*Additional context**
The same problem occurs on the OxygenOS rom (OxygenOS-AB-10-20191010), this is the log for that rom:
[logcat_oos.log](https://github.com/erfanoabdi/ErfanGSIs/files/3721385/logcat_oos.log)

Thanks for your attention :)

erfanoabdi, 2019-10-13:
need log from boot time

depauperado, 2019-10-13:
Hi, 

Thanks for your quick response

This is the file located on the /data/local/tmp/logs directory:

[log_MIUI-AB-10-20191012.log](https://github.com/erfanoabdi/ErfanGSIs/files/3722147/log_MIUI-AB-10-20191012.log)

I hope it helps you

Thank you

erfanoabdi, 2019-10-13:
wew, it's super wired error, no idea bro
does this camera works on phh Q gsis?

depauperado, 2019-10-14:
I have tried some Q gsis from phh and didn't find any that boot.

I guess I got lucky that your most recent gsis boot on my device =(

thanks for you attention, should i close this issue?