erfanoabdi/ErfanGSIs #96: <Android 10> <Mi 8 Lite (Platina)> [Bug name: Camera is dead]

FuzzySnuk, 2019-10-16:
**Describe the bug**
'Camera is not working in Pixel Q gsi. `

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Log file:[Qcamera.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3736323/Qcamera.txt)

(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [Platina]
 - Vendor: [MIUI  Global Stable 10.3.5.0]
 - Version [If you are asking for Vendor veriosn than it's "Pie" If you are asking GSI version than it's "Q" aka "Android 10"]

**Additional context**
Add any other context about the problem here.

MacTavishAO, 2019-10-21:
Why do you have whyred fingerprint ?

FuzzySnuk, 2019-10-21:
Because it's a GSI ported with erfan tools from whyred

zainarbani, 2019-11-29:
E HAL     : dlopen failed: cannot locate symbol "_ZNK7minikin6Layout10getAdvanceEv" referenced by "/vendor/lib/libMiCameraHal.so"...

https://github.com/xiaomi-sdm660/android_vendor_xiaomi_wayne/commit/6c411dc80cf4cb3529aa5f4efe5d5a4169eaf4bf