erfanoabdi/ErfanGSIs #99: <9.0 MIUI11 Global Stable> <Samsung S7 Edge Exynos> [Black Screen]

modaifallah, 2019-10-22:
**Describe the bug**
The phones boots normally, I can use the device for long time, download apks and browse Internet. Looks very stable than the Chinese Beta Pie MIUI 11. The problem is that once I try to pull status bar down or lock the device, the phone turns into black screen. The same happens if I lock the device with power button and unlock again. Not fixed by a reboot, need to format data partition and start over. 
While in black screen, long pressing power button gives me power menu, and I can hear UI sounds. Tried deleting Provision and MiuiHome apps, and also /vendor/overlays folder. 

Also as you guys are checking the logs, brightness in this MIUI can be changed, but the range is wrong, can this be fixed too? 
**To Reproduce**
Steps to reproduce the behavior:
Pull status bar down to open quick settings/notifications, or lock the device. 

**Expected behavior**
Black Screen. 
 
**Screenshots*

**Logs**
[log.txt](https://github.com/erfanoabdi/ErfanGSIs/files/3753859/log.txt)
)

**Smartphone (please complete the following information):**
 - Device: Samsung S7 Edge Exynos 
 - Vendor: Project Pizza Treble Vendor (only vendor we have, boots AOSP "bugfree" and old MIUI Bega very good. 
 - Version: Pie

shivvvvvv, 2019-10-22:
Delete dpmservice app. Its in system/priv-app or system/app.

modaifallah, 2019-10-22:
@shivvvvvv Didn’t work, same behavior

kiss04, 2019-11-25:
good