erfanoabdi/ErfanGSIs #101: <Android version 9 > <Device code name RMX 1921> Bootloop on animation

nandha-2point0, 2019-11-10:
@**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

nandha-2point0, 2019-11-10:
Device: rmx1921
Vendor: stock
Version :9

closed 2019-11-10 by nandha-2point0

4PERTURE, 2019-11-10:
@nandha-2point0 Great description