erfanoabdi/ErfanGSIs #102: <Android 10> <oneplus3> [New Pixel GSI A/B roms not boot]

lufog, 2019-11-14:
**Describe the bug**
All Pixel A/B 10 GSIs that older than **20190926** stuck on "POWERED BY ANDROID"

**To Reproduce**
1. Flash latest **Pixel-AB-10-20191107-ErfanGSI.img**
2. Reboot phone
4. Stuck on "1+ POWERED BY ANDROID" screen

**Expected behavior**
Normal boot

**Logs**
Log file: [log.zip](https://github.com/erfanoabdi/ErfanGSIs/files/3844454/log.zip)

**Smartphone:**
 - Device: OnePlus 3T
 - Vendor: [LineageOS 16.0 Treble (system-as-root)](https://forum.xda-developers.com/oneplus-3/oneplus-3--3t-cross-device-development/treble-lineageos-15-1-treble-oneplus-3-t3830455)
 - Version LineageOS Pie, GSI Android 10

**Additional context**
Pixel A/B 10 GSIs **20190904** and **20190926** are works, they are buggy but works.

zainarbani, 2019-11-29:
F linker  : CANNOT LINK EXECUTABLE "/vendor/bin/hw/android.hardware.audio@2.0-service": library "libqinder.so" not found

haxed module ??

erfanoabdi, 2020-02-19:
fix merged https://github.com/erfanoabdi/ErfanGSIs/commit/71432169c867e35f67991562acbde800fa844182

closed 2020-02-19 by erfanoabdi