erfanoabdi/ErfanGSIs #106: <Android version> <Device code name> [Bug name Ex. Bootloop on animation]

marious1997, 2019-12-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logs**
Add logs to help your problem.
Log file: /data/local/tmp/logs/log
(Bug reports without logs will be rejected)

**Smartphone (please complete the following information):**
 - Device: [e.g. Tissot]
 - Vendor: [e.g. Lineage 16]
 - Version [e.g. Pie]

**Additional context**
Add any other context about the problem here.

TheGabrielHoward, 2019-12-21:
@marious1997 nice issue

MacTavishAO, 2019-12-23:
There isn't anything useful. Closed

closed 2019-12-23 by MacTavishAO