erfanoabdi/ErfanGSIs #134: about high RAM used

Nightmare-MY, 2020-05-16:
when i  use mkuserimg_mke2fs.sh to build new ext img,i found that when i add args -s,i'll get a new sparse img,bug high RAM consumption while building new img,iso i doesn't use -s args,It works very well,and i'll get new rimg,the next i use img2simg tool to convert my new rimg to simg.
it looks like a good solution,maybe i can Submit a pr,Any answers, thank you.

erfanoabdi, 2020-05-16:
it's aosp script and -s is what aosp uses to make images https://android.googlesource.com/platform/system/extras/+/master/ext4_utils/mkuserimg_mke2fs.py#64
i didn't notice any high ram usage here

Nightmare-MY, 2020-05-16:
thank you for you replay,But it does actually consume a lot of RAM,I guess maybe because e2fsdroid is working as a sparse mirror it leads to a higher memory footprint,I modified your script
```
if [ "$5" == "--old" ]; then
    if [ "$outputtype" == "Aonly" ]; then
        sudo $make_ext4fs -T 0 -S $fcontexts -l $syssize -L system -a system -s "$output" "$systemdir/system"
    else
        sudo $make_ext4fs -T 0 -S $fcontexts -l $syssize -L / -a / -s "$output" "$systemdir/"
    fi
else
    if [ "$outputtype" == "Aonly" ]; then
        sudo $toolsdir/mkuserimg_mke2fs.sh "$systemdir/system" "$output" ext4 system $syssize -T 0 -L system $fcontexts
    else
        sudo $toolsdir/mkuserimg_mke2fs.sh "$systemdir/" "$output" ext4 / $syssize -T 0 -L / $fcontexts
    fi
    mv "$output" system.rimg
    img2simg system.rimg "$output"
fi
```
and in low ram device works very well,
finally I'm about to port your tools to an android device
and Looks like a success,the only problem about e2fsdroid,i doesn't has a static e2fsdroid,i only has dynamic linked e2fsdroid,it will rely some /lib/so,I was able to use it successfully but it was very cumbersome,I've been trying to compile static e2fsdroid for a long time,but it never worked

Nightmare-MY, 2020-05-16:
I'm Chinese,sorry about my English.

yukosky, 2020-06-26:
Try using swap (6/8gb)

Nightmare-MY, 2020-06-27:
> Try using swap (6/8gb)

i means can repack rimg to resolve this problem

yukosky, 2020-06-27:
> > Try using swap (6/8gb)
> 
> i means can repack rimg to resolve this problem

Aaah, okay

closed 2021-08-17 by Nightmare-MY