erfanoabdi/ErfanGSIs #141: <Android 11 Beta> <Lima> [Device does not start / stuck in bootloop]

RickHVBN, 2020-06-25:
**Describe the bug**
My Moto One Macro it started the GSIs of Android 11 Developer Preview, but now with the new GSIs of Android 11 Beta, the device will not start!

**To Reproduce**
Steps to reproduce the behavior:
1. Install the latest GSI from Android R Developer Preview, and install the new loved version of Android 11 Beta!
2. See that GSI does not start in its newest version

**Expected behavior**
Since the device supports Project Treable, its system is 64-bit, and has A / B partitions, I expected that the Beta version of Android 11 runs on the device!

**Screenshots**
That's basically it, the device does not start on the newest version of the Android 11 GSIs

**Logs**
There is no problem related to installation, everything just happens!

**Smartphone (please complete the following information):**
 - Device: Motorola One Macro
 - Vendor: Pixel GSI
 - Version: Android 11 Beta

**Additional context**
I hope there is a correction, since the old version normally runs on the device!

nitanmarcel, 2020-07-19:
Try using the official gsi + permissiver

yukosky, 2020-07-19:
Vendor Pixel GSI?¿