erfanoabdi/ErfanGSIs #184: Generate image error and file empty (mke2fs: invalid blocks)

abda11ah, 2022-01-01:
I tried to generate a GSI.
I'm running manjaro linux and I installed all the dependencies + java jre 11 because I got errors on Java17 during compilation.

But there is a persistent error and the generated GSI file size is 0 bytes.

Here is the log :

> ./url2GSI.sh https://github.com/PotatoDevices/device_xiaomi_raphael/releases/download/4.2.3/potato_raphael-11-20210909-dumaloo.v4.2.3+20.Crispy.zip Generic AB
> Updating tools...
> Récupération du sous-module prebuilt/vendor_vndk
> Récupération du sous-module tools/ROM_resigner
> Déjà à jour.
> HEAD est maintenant à e851538 Use gnu coreutils on Darwin
> Downloading firmware to: /home/client/ErfanGSIs/input/24609_FIRMWARE.tgz
> 
> 01/01 09:51:23 [NOTICE] Downloading 1 item(s)
> 
> 01/01 09:51:24 [NOTICE] CUID#7 - Redirecting to https://objects.githubusercontent.com/github-production-release-asset-2e65be/240904495/381e4ac5-4e88-4b72-8a60-6625aef32daf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220101%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220101T085123Z&X-Amz-Expires=300&X-Amz-Signature=a53a34b501a88a8a0746d1fa8c0f603dcf09589c14e3e05f5c05d42d874750b3&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240904495&response-content-disposition=attachment%3B%20filename%3Dpotato_raphael-11-20210909-dumaloo.v4.2.3%2B20.Crispy.zip&response-content-type=application%2Foctet-stream
> 
> 01/01 09:51:24 [NOTICE] CUID#11 - Redirecting to https://objects.githubusercontent.com/github-production-release-asset-2e65be/240904495/381e4ac5-4e88-4b72-8a60-6625aef32daf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220101%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220101T085124Z&X-Amz-Expires=300&X-Amz-Signature=bce2e53bc5a7290bb8161b40592431799032263bb4d39b1e5a633147c9d26a01&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240904495&response-content-disposition=attachment%3B%20filename%3Dpotato_raphael-11-20210909-dumaloo.v4.2.3%2B20.Crispy.zip&response-content-type=application%2Foctet-stream
> 
> 01/01 09:51:24 [NOTICE] CUID#12 - Redirecting to https://objects.githubusercontent.com/github-production-release-asset-2e65be/240904495/381e4ac5-4e88-4b72-8a60-6625aef32daf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220101%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220101T085124Z&X-Amz-Expires=300&X-Amz-Signature=bce2e53bc5a7290bb8161b40592431799032263bb4d39b1e5a633147c9d26a01&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240904495&response-content-disposition=attachment%3B%20filename%3Dpotato_raphael-11-20210909-dumaloo.v4.2.3%2B20.Crispy.zip&response-content-type=application%2Foctet-stream
> 
> 01/01 09:51:24 [NOTICE] CUID#9 - Redirecting to https://objects.githubusercontent.com/github-production-release-asset-2e65be/240904495/381e4ac5-4e88-4b72-8a60-6625aef32daf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220101%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220101T085124Z&X-Amz-Expires=300&X-Amz-Signature=bce2e53bc5a7290bb8161b40592431799032263bb4d39b1e5a633147c9d26a01&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240904495&response-content-disposition=attachment%3B%20filename%3Dpotato_raphael-11-20210909-dumaloo.v4.2.3%2B20.Crispy.zip&response-content-type=application%2Foctet-stream
> 
> 01/01 09:51:24 [NOTICE] CUID#10 - Redirecting to https://objects.githubusercontent.com/github-production-release-asset-2e65be/240904495/381e4ac5-4e88-4b72-8a60-6625aef32daf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220101%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220101T085124Z&X-Amz-Expires=300&X-Amz-Signature=bce2e53bc5a7290bb8161b40592431799032263bb4d39b1e5a633147c9d26a01&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240904495&response-content-disposition=attachment%3B%20filename%3Dpotato_raphael-11-20210909-dumaloo.v4.2.3%2B20.Crispy.zip&response-content-type=application%2Foctet-stream
> 
> 01/01 09:51:24 [NOTICE] CUID#12 - Redirecting to https://objects.githubusercontent.com/github-production-release-asset-2e65be/240904495/381e4ac5-4e88-4b72-8a60-6625aef32daf?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20220101%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20220101T085124Z&X-Amz-Expires=300&X-Amz-Signature=bce2e53bc5a7290bb8161b40592431799032263bb4d39b1e5a633147c9d26a01&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=240904495&response-content-disposition=attachment%3B%20filename%3Dpotato_raphael-11-20210909-dumaloo.v4.2.3%2B20.Crispy.zip&response-content-type=application%2Foctet-stream
> [#b28a3f 586MiB/586MiB(99%) CN:1 DL:14MiB]                                                                                                  
> 01/01 09:51:56 [NOTICE] Téléchargement terminé: /home/client/ErfanGSIs/input/24609_FIRMWARE.tgz
> 
> Résultats du téléchargement:
> gid   |stat|avg speed  |path/URI
> ======+====+===========+=======================================================
> b28a3f|OK  |    21MiB/s|/home/client/ErfanGSIs/input/24609_FIRMWARE.tgz
> 
> Légende du statut:
> (OK): téléchargement terminé.
> Déjà à jour.
> Mise à jour d02128d..1a1ce49
> Fast-forward
>  README.md      |   7 ++----
>  ozipdecrypt.py | 138 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++----------------------------------------------------
>  2 files changed, 79 insertions(+), 66 deletions(-)
> Déjà à jour.
> Create Temp and out dir
> Extracting firmware on: /home/client/ErfanGSIs/working
> Aonly OTA detected
> Converting brotli system dat to normal
> Extracting system
> Create Temp dir
> Making copy of source rom to temp
> Resigning to AOSP keys
> Patching started...
> Using device specific overlays is not supported in this rom. Skipping...
> Creating Image: Generic-AB-11-20220101-ErfanGSI.img
> ERROR:root:b"mke2fs 1.45.4 (23-Sep-2019)\nmke2fs: invalid blocks '413093.0998535156' on device '/home/client/ErfanGSIs/output/Generic-AB-11-20220101-ErfanGSI.img'\n"
> Remove Temp dir
> Create Temp dir
> Making copy of source rom to temp
> Resigning to AOSP keys
> Patching started...
> Using device specific overlays is not supported in this rom. Skipping...
> Generating environ.rc
> Creating Image: Generic-Aonly-11-20220101-ErfanGSI.img
> ERROR:root:b"mke2fs 1.45.4 (23-Sep-2019)\nmke2fs: invalid blocks '414297.4499511719' on device '/home/client/ErfanGSIs/output/Generic-Aonly-11-20220101-ErfanGSI.img'\n"
> Remove Temp dir
> Porting Generic GSI done on: /home/client/ErfanGSIs/output

Firmware is correctly downloaded, filesize is about 586 MB

abda11ah, 2022-01-05:
Fixed.
I had to clone an older fork from "VegaBobo" to make it works.

[https://github.com/VegaBobo/ErfanGSIs](url)

tosasitill, 2023-02-05:
hey bro I solved this problem just now
the solvution is use --old to use mke4fs to creat img