erfanoabdi/ErfanGSIs #190: ERROR:root:b"mke2fs 1.45.4 (23-Sep-2019)\nmke2fs: invalid blocks '793524.8999023438' on device

Dachoni03, 2022-05-28:
When I try to make a Generic GSI this Problem Appears. Does anyone Can Help Me?

Erox1024, 2022-08-13:
I also met this problem when making Android 10 GSIs. But when I make Android 9 GSIs, it went on fine.

mi20210630, 2023-01-08:
ERROR:root:b"mke2fs 1.45.4 (23-Sep-2019)\nmke2fs: invalid blocks '1228195.5' on device '/home/runner/work/ErfanGSIs-online/ErfanGSIs-online/ErfanGSIs/output/Generic-AB-11-20230108-ErfanGSI.img'\n"
Is this full error? I made this again and can't fix it (ubuntu20.04environment).

tosasitill, 2023-02-05:
hey bro I solved this problem just now
the solvution is use --old to use mke4fs to creat img

narasimha00, 2023-04-11:
> hey bro I solved this problem just now the solvution is use --old to use mke4fs to creat img
Please tell me in detail !!

tosasitill, 2023-04-11:
> > hey bro I solved this problem just now the solvution is use --old to use mke4fs to creat img
> > Please tell me in detail !! about how do I use --old ??

Look here :https://github.com/tosasitill/erfanSGSI/commit/870d0df3edc1047dc9017b38c5e6c84c0165d195